mruijter/lora_transceiver #1: I try this code in sx1276 but I still can`t change sf

weichingLo, 2019-03-20:
Hi, I have tried this code in sx1276 and it worked in sf7 868.1mhz bw125khz
but I tried to change sf to 12 and then I can`t receive any signal.
config.txt is the same as yours.
Or I should change it in REG_MODEM_CONFIG2?

mruijter, 2019-03-20:
Hi weichingLo,

When you set : debug=enabled do you see that lora_transceiver logs SX1276 detected at startup?

```
Verbose level: 1
Selected bandwidth: BW125
Selected spreading factor: SF8
CRC checks: disabled
Selected coding rate: CR4_6
Debugging: enabled
Send / receive packets at:  SF8 on 869.550000 Mhz.
Fifo /dev/shm/receive_fifo is now open
Fifo /dev/shm/send_fifo is now open
SX1276 detected, starting.
```

Are both nodes running lora_transceiver or is the peer node running the original broken dragino code? As far as I can tell lora_transceiver does switch to a different SF rate.
You can see that the time it takes to transmit a message changes dramatically when selecting a higher SF.

For example SF8 vs SF12:
```
SF8:
Sending 31 bytes.
..
send: 31 bytes in 140157 microseconds

SF12:
Sending 31 bytes.
send: 31 bytes in 1843664 microseconds
```

weichingLo, 2019-03-20:
hi, I can see sx1276 in debug mode and only the receiving node running lora_transceiver.
the other Lora node is my icon... 
I had switch all the sf 7-12 in both device but it still work in both in sf 7
I tried your code for looking a solution for another way to deal with the problem that Lora range is only in 200 meters. thanks for your answer and I will try to force it to sf 12 in REG_MODEM_CONFIG2

weichingLo, 2019-03-20:
Send / receive packets at:  SF8 on 869.550000 Mhz.
It doesn`t show on my RPI

closed 2019-04-29 by mruijter