mruijter/lora_transceiver #2: Huge packet loss

Michael-List, 2019-04-24:
Hi,

I have an Arduino MKR 1300 which sends weather data ([see this code](https://github.com/Michael-List/arduino_lora_weatherstation)) via lora.
If i use my RaspberryPi with the Dragino hat with the original dragino code the RaspberryPi receives nearly every message. If i use your code, I only receive a few messages -> over 80 % loss. The distance between the RaspberryPi and the Arduino is 1m. The frequency, spreading factor, bandwith, coding rate and crc settings are the same on both. Do you have an idea what could be the problem?

Greetings

mruijter, 2019-04-24:
Are your messages coded as fixed blocksize? The example config assumes 32 byte fixed length messages. A mismatch could explain the problem.

Michael-List, 2019-04-25:
Yep i set it to a fixed size. Tested different blocksizes also. Did you include another filter or configured something different like in the original repo?

mruijter, 2019-04-25:
The main difference between the original code and this code is that changing speed and SF now actually works.

You can easily diff the two.
The PI is the receiver?

Michael-List, 2019-04-25:
Ok. I will make a diff and look at this details. Yes the pi is the receiver

mruijter, 2019-04-25:
wfd = open_create_fifo(LORASEND_FIFO, O_RDWR | O_NONBLOCK);

The data is written to the pipe using non blocking io. So reading it when it arrives is required or the data will be discarded.

mruijter, 2019-04-27:
I can reproduce the problem. I created a numbered packet stream and I am missing packets after correctly receiving a batch of them. Working on a fix..

Michael-List, 2019-04-27:
Ok that's good news. I was about to make some tests tomorrow. But with this news I'm waiting for your fix.

Michael-List, 2019-04-29:
I found the problem. If i delete the line `writeReg(REG_OPMODE, OPMODE_LORA); // Set the register enabled for Lora.` I receive every packet I send.

mruijter, 2019-04-29:
Thanks Michael. I read the sx1276 manual twice. And I fail to see why this code was problematic. However the original code did not have it and I set the opmode to lora later on. So it was redundant at best. I’ve removed it.

Michael-List, 2019-04-29:
Thanks for your help mruijter. This issue refers to commit a2f1da5b3c2dbcf3d159d39bf5fb4b3ad6ea96fc

closed 2019-04-29 by Michael-List