mruijter/lora_transceiver #4: Stop sending after a while

perkasajob, 2020-01-14:
Hi mruijter,
I tried to send out regularly, using "while true; do echo "hello lora" >/dev/shm/send_fifo; sleep 1; done"
on the receiver side arduino dragino shield is used. but seems after a while it stop sending, until the lora.service is restarted, but then it stops again after a while (approx. 10x sending), 
my config is: SF7, BW125, blocksize 3 (tried also 32), CR4_8, CRC=enabled., on the Arduino side, is configured the identical.

How could I debug this ? where to look for this kind of problem?
Thank You, Regards.

mruijter, 2020-01-14:
Can you check what happens when the blocksize matches the message size?
Hello lora = 10 long?

The config file allows you to enable debug. When you start the program directly without systemd it may explain what it is doing.

perkasajob, 2020-01-14:
the following it the debug :
At first :  
pi@raspberrypi:~/lora_transceiver $ sudo ./lora_transceiver 
------------------------------------
Verbose level: 1
Selected bandwidth: BW125
Selected spreading factor: SF7
CRC checks: enabled
Selected coding rate: CR4_8
Debugging: enabled
Send / receive packets at:  SF7 on 915.000000 Mhz.
Fifo /dev/shm/receive_fifo is now open
Fifo /dev/shm/send_fifo is now open
SX1276 detected, starting.
writeReg:
81 08 
writeReg:
86 E4 
writeReg:
87 C0 
writeReg:
88 00 
writeReg:
B9 34 
writeReg:
A6 04 
writeReg:
9D 78 
writeReg:
9E 74 
writeReg:
9F 08 
writeReg:
A3 80 
writeReg:
A2 40 
writeReg:
A4 FF 
writeReg:
8D 0A 
writeReg:
8C 23 
BW = 0, coding_rate = 0, implicit_header_mode = 0
opmodeLora 80 -> 1
writeReg:
81 88 

then the last 2 transmit messages
------------------
Sending 10 bytes.
6C 6F 20 6C 6F 72 61 0A 68 65 
writeReg:
C0 F0 
writeReg:
92 FF 
writeReg:
91 F7 
writeReg:
8E 00 
writeReg:
8D 00 
writeReg:
A2 0A 
writeReg:
81 8B 
send: 10 bytes in 36914 microseconds
------------------
Sending 10 bytes.
6C 6C 6F 20 6C 6F 72 61 0A 68 
writeReg:
C0 F0 
writeReg:
92 FF 
writeReg:
91 F7 
writeReg:
8E 00 
writeReg:
8D 00 
writeReg:
A2 0A 
writeReg:
81 8B 
send: 10 bytes in 36592 microseconds

For my novice eyes, they are fine, but it just stops.

mruijter, 2020-01-14:
I'll repeat the test with the same settings. I assume that frequency that you use is legal where you are located. I have tested only in the 868Mhz range but I don't think that will be relevant in this case. Not sure what causes transmissions to stop in your case.

closed 2020-01-15 by perkasajob