mruijter/lora_transceiver #6: Lora Transciever does not work with 64 Bit RPi OS?

mxsrv, 2021-06-23:
Hello,

I am running RaspberryPi OS 64 Bit and wanted to establish a connection between 2 LoRa hats. I am using 64 Bit because I am running a backend with mongodb and want to use more than 2GB. With 32bit I am limited to 2GB of DB storage. After running `make` i got this:

`g++ -I ./include lora_transceiver.c lib_log.c lib_str.c lib_cfg.c -lwiringPi -o lora_transceiver`
`/usr/bin/ld: skipping incompatible /usr/lib/gcc/aarch64-linux-gnu/8/../../../../lib/libwiringPi.so when searching for -lwiringPi`
`/usr/bin/ld: skipping incompatible /lib/../lib/libwiringPi.so when searching for -lwiringPi`
`/usr/bin/ld: skipping incompatible /usr/lib/../lib/libwiringPi.so when searching for -lwiringPi`
`/usr/bin/ld: skipping incompatible /usr/lib/gcc/aarch64-linux-gnu/8/../../../libwiringPi.so when searching for -lwiringPi`

Note: I first installed `lwiringpi` over `apt-get` but `make` failed completely and I then installed a 64 Bit version of it.

While running `make install` I then got this:

`install -m 755 lora_transceiver /usr/bin`
`install -m 644 lora.service /lib/systemd/system`
`install -D -b -m 644 lora.conf-example /etc/lora/lora.conf`
`systemctl daemon-reload`
`systemctl restart lora`
`systemctl status lora`
`● lora.service - Starts the lora transceiver.` Loaded: loaded (/lib/systemd/system/lora.service; disabled; vendor preset: enabled)
   Active: activating (auto-restart) (Result: exit-code) since Wed 2021-06-23 12:37:42 CEST; 20ms ago
  Process: 12983 ExecStart=/usr/bin/lora_transceiver (code=exited, status=1/FAILURE)
 Main PID: 12983 (code=exited, status=1/FAILURE)
make: *** [Makefile:20: install] Error 3`

josemqz, 2023-09-07:
What version of RaspberryPi OS are you running?  
I'm using version 11 (Bullseye) and installed WiringPi by running

josemqz, 2023-09-07:
What version of RaspberryPi OS are you running?
I'm using version 11 (Bullseye) 64-bit, and installed WiringPi by running `sudo apt install wiringpi`.